Livox-SDK/livox_ros_driver2 #184: Default IP address in config may not match new MID360 sensors

tedsamu, 2025-04-25:
I recently got a Livox MID360 working with the ROS 2 drivers. The main issue I ran into was that the sensor's default IP address was 192.168.1.120, while the example config uses 192.168.1.12.

Is this a typo in the default config, or do some devices come with different default IPs?

It would be great if the example launch/config files worked out-of-the-box with a brand new sensor, or if the docs mentioned this potential mismatch.

Diff i use to get it working:
```
diff --git a/config/MID360_config.json b/config/MID360_config.json
index e3df5d1..950fed3 100644
--- a/config/MID360_config.json
+++ b/config/MID360_config.json
@@ -25,7 +25,7 @@
   },
   "lidar_configs" : [
     {
-      "ip" : "192.168.1.12",
+      "ip" : "192.168.1.120",
       "pcl_data_type" : 1,
       "pattern_mode" : 0,
       "extrinsic_parameter" : {
```

georgflick, 2025-04-28:
The default IP is dependent on the serial number from the sensor.
Copied from the manual:
```
All Livox Mid-360 LiDAR sensors are set to static IP address mode by default with an IP
address of 192.168.1.1XX (XX stands for the last two digits of the Livox Mid-360 LiDAR sensor’s serial
number)
```

tedsamu, 2025-04-28:
Thanks for the clarification!

That makes sense, I must have missed that detail in the manual.
Maybe it would be helpful to mention this in the README or near the example configs, just so new users don't get tripped up?

closed 2025-04-28 by tedsamu