Livox-SDK/livox_ros_driver2 #199: Inquiry regarding the configuration of the region of interest (ROI) of the MID-360 sensor....

wookbin, 2025-06-06:
Hello,
I have a question regarding the Region of Interest (ROI) configuration for the MID-360 sensor.

Due to the design of the robot platform I am using, I am unable to utilize the full 360-degree field of view of the sensor. The sensor is mounted at a specific position on the lower front side of the robot, and approximately 90 degrees of the rear view is obstructed. Therefore, I would like to limit the sensor's detection area to 270 degrees horizontally.

As far as I know, the MID-360 provides a 360-degree horizontal and 59-degree vertical field of view.
Is there a way to configure a specific ROI for this sensor? For example, is it possible to limit the horizontal field of view to 270 degrees and the vertical field of view to 30 degrees?

With other sensors, I have seen that parameters such as range_min and range_max can be set in the launch file to control this. However, I was not able to find such parameters for the MID-360.

I would appreciate any guidance or suggestions regarding this matter.

wookbin, 2025-06-11:
Thanks for the answer.

'passes through all points where x value is between 0 and 50' -> What are the units of the values ​​you are talking about here? Is it simply a threshold, or degrees or radians?

FelixCooper1026, 2025-06-23:
Mid-360 supports ROI configuration, simply follow the steps below:
1. Connect Mid-360 to your PC
2. Download LivoxViewer2 from the official Livox website [https://www.livoxtech.com/downloads](url)
3. Open LivoxViewer2 and go to "Device settings" page
4. Find "FOV enable" and switch to "FOV0/1 enable" or "Both enable"
5. Configure ROI by changing the values of "FOV Cfg0" and "FOV Cfg1" (Horizontal 0°~360°, Vertical -7°~52°）
6. Click "Confirm" to save
7. Launch livox ros driver2 and check the results

[image]

wookbin, 2025-06-27:
I recently found out that there is 'LivoxViewer2'. 
I solved it with this program and applied it. ^^