Livox-SDK/livox_ros_driver2 #206: Slower lidar publish rate than expected

Astro-Coco, 2025-07-17:
Hi, 

I'm using the mid360 lidar to run some slam algorithms. 
While /livox/imu publishes at 200 Hz steady, the /livox/lidar topic seem to only publish at 75% the speed of what's set in the launch file (roughly 7.5 Hz instead of 10 Hz). I have check cpu load, temp, and udp packet drop, but could not find anything meaningful. This happens when nothing else is running. The driver is running inside a docker container on a raspberry pi 5. The docker's base image is ros2 humble on ubuntu 22.04. 

Rate seems to drop even more when I start the rest of the pipeline. 

I have narrowed down the problem to the driver, altough the issue might also be related to hardware.

Do  you know about this issue and what might be causing it?

georgflick, 2025-09-03:
> Hi,
>
> I'm using the mid360 lidar to run some slam algorithms. While /livox/imu publishes at 200 Hz steady, the /livox/lidar topic seem to only publish at 75% the speed of what's set in the launch [...]

if it gets worse, when you are starting the rest of the pipeline, then maybe the raspberry pi 5 is just not powerful enough? Could be worth to try out intra process communication or zenoh.

TanayK07, 2026-10-15:
one check that splits this: echo a single message and look at how much time it covers. with the CustomMsg format the last point's offset_time gives the span (in PointCloud2 mode, the per-point timestamp field). if each message holds ~133 ms of points the driver is batching longer frames, if it holds ~100 ms then whole frames are going missing. also on a pi 5, ros2 topic hz on a full mid-360 cloud can read low by itself, so it's worth confirming the driver's real publish rate from inside. i wrote ros2_pulse for that: LD_PRELOAD it into the driver launch and it logs how often the driver actually publishes per topic for the whole run, no subscriber, which would also show @ClemensLinnhoff whether the drop at 4-5 min is in the driver or downstream. it only counts messages, so it won't tell you why the driver slows, just where. https://github.com/TanayK07/ros2_pulse