Livox-SDK/livox_ros_driver2 #208: Unable to Receive Data from HAP TX  - bind failed; Failed to init livox lidar sdk.

ehsan-javanmardi, 2025-07-30:
I am using a HAP TX LiDAR and trying to receive its data using ROS 2. However, I am unable to get any data.

Here is a summary of the issue:
When I run **Livox Viewer 2**, I can see the data stream, and the HAP sensor turns on (the fan starts, and the device makes a noise indicating it's active).
But when I run the ROS 2 launch command, 
`ros2 launch livox_ros_driver2 rviz_HAP_launch.py `

The sensor remains silent (no fan or noise), and I encounter the following error:
```
[INFO] [launch]: All log files can be found below /home/ehsan/.ros/log/2025-07-30-21-21-33-098139-ehsan-ada4500-39935
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [livox_ros_driver2_node-1]: process started with pid [39936]
[INFO] [rviz2-2]: process started with pid [39938]
[livox_ros_driver2_node-1] [INFO] [1753878093.138577944] [livox_lidar_publisher]: Livox Ros Driver2 Version: 1.2.4
[livox_ros_driver2_node-1] [INFO] [1753878093.138842684] [livox_lidar_publisher]: Data Source is raw lidar.
[livox_ros_driver2_node-1] [INFO] [1753878093.138865928] [livox_lidar_publisher]: Config file : /home/ehsan/workspace/livox/livox_ws2/install/livox_ros_driver2/share/livox_ros_driver2/launch_ROS2/../config/HAP_config.json
[livox_ros_driver2_node-1] LdsLidar *GetInstance
[livox_ros_driver2_node-1] config lidar type: 8
[livox_ros_driver2_node-1] successfully parse base config, counts: 1
[livox_ros_driver2_node-1] bind failed
[livox_ros_driver2_node-1] Failed to init livox lidar sdk.
[livox_ros_driver2_node-1] [ERROR] [1753878093.140501874] [livox_lidar_publisher]: Init lds lidar fail!
[rviz2-2] [INFO] [1753878093.534087881] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-2] [INFO] [1753878093.534153850] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
[rviz2-2] [INFO] [1753878093.568615094] [rviz2]: Stereo is NOT SUPPORTED
```

Here is the HAP_config.json 
```
{
  "lidar_summary_info" : {
    "lidar_type": 8
  },
  "HAP": {
    "lidar_net_info" : {
      "cmd_data_port": 56000,
      "push_msg_port": 0,
      "point_data_port": 57000,
      "imu_data_port": 58000,
      "log_data_port": 59000
    },
    "host_net_info" : {
      "cmd_data_ip" : "192.168.1.13",
      "cmd_data_port": 56000,
      "push_msg_ip": "",
      "push_msg_port": 0,
      "point_data_ip": "192.168.1.13",
      "point_data_port": 57000,
      "imu_data_ip" : "192.168.1.13",
      "imu_data_port": 58000,
      "log_data_ip" : "",
      "log_data_port": 59000
    }
  },
  "lidar_configs" : [
    {
      "ip" : "192.168.1.100",
      "pcl_data_type" : 1,
      "pattern_mode" : 0,
      "extrinsic_parameter" : {
        "roll": 0.0,
        "pitch": 0.0,
        "yaw": 0.0,
        "x": 0,
        "y": 0,
        "z": 0
      }
    }
  ]
}
```

What I have tried:

- I confirmed the HAP configuration. My host PC IP is set to 192.168.1.13, which matches the HAP network settings.
- I disabled the firewall to rule out any network filtering.
- I successfully ran the Livox SDK 2 quick start sample, and it worked fine. (The device turns active)
- I also tried changing the HAP broadcast code (BD code) in the launch file:
`# cmdline_bd_code = 'livox0000000001' `
to 
`cmdline_bd_code = '5CWD244N4109AS'`
Despite all this, I still cannot receive data using ROS 2, and the HAP sensor does not appear to start properly.

Request:
Any suggestions or guidance would be appreciated. It seems like the device does not initialize correctly under ROS 2, even though it works with Livox Viewer 2 and the SDK directly.

rss-hiverobots, 2025-09-03:
Hello, did you manage to solve this? If so, how?
Thanks

yogachuan, 2025-09-18:
你需要同步修改两个位置的hap配置文件，一个位于xxxx/ws_livox/src/livox_ros_driver2/config/HAP_config.json，另一个位于xxxx/ws_livox/install/livox_ros_driver2/share/livox_ros_driver2/config/HAP_config.json

real-lsy, 2025-10-17:
@yogachuan It has been bothering me for many days,finally solved it.Thank u!