Livox-SDK/livox_ros_driver2 #220: Compile livox_ros_driver2，got an error

shisuhao, 2025-09-16:
Starting >>> livox_ros_driver2ngoing][7h 2min 24.4s] [1/3 complete]             
0;colcon build [1/3 done] [1 ongoing][7h 2min 24.5s] [1/3 complete] [livox_ros_d
--- stderr: livox_ros_driver2                                                   
Traceback (most recent call last):                                              
  File "/usr/local/lib/python3.9/site-packages/colcon_core/executor/__init__.py"
, line 91, in __call__                                                          
    rc = await self.task(*args, **kwargs)                                       
  File "/usr/local/lib/python3.9/site-packages/colcon_core/task/__init__.py", li
ne 93, in __call__                                                              
    return await task_method(*args, **kwargs)                                   
  File "/usr/local/lib/python3.9/site-packages/colcon_ros/task/ament_cmake/build
.py", line 59, in build                                                         
    rc = await extension.build(                                                 
  File "/usr/local/lib/python3.9/site-packages/colcon_cmake/task/cmake/build.py"
, line 92, in build                                                             
    project_name = get_variable_from_cmake_cache(                               
  File "/usr/local/lib/python3.9/site-packages/colcon_cmake/task/cmake/__init__.
py", line 246, in get_variable_from_cmake_cache                                 
    lines = _get_cmake_cache_lines(path)                                        
  File "/usr/local/lib/python3.9/site-packages/colcon_cmake/task/cmake/__init__.
py", line 265, in _get_cmake_cache_lines                                        
    content = h.read()                                                          
  File "/usr/lib64/python3.9/codecs.py", line 322, in decode                    
    (result, consumed) = self._buffer_decode(data, self.errors, final)          
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xe0 in position 47212: inva
lid continuation byte                                                           
---                                                                             
Failed   <<< livox_ros_driver2 [7.76s, exited with code 1]                      
                                         2min 32.2s] [2/3 complete] [1 failed]  
Summary: 1 package finished [7h 2min 32s]                                       
  1 package failed: livox_ros_driver2                                           
  2 packages had stderr output: VTK livox_ros_driver2                           
  1 package not processed                                                       
'utf-8' codec can't decode byte 0xe0 in position 47212: invalid continuation byt
e 

How can i solve it？
Please give me some suggestion.

shisuhao, 2025-09-16:
@Livox-louie @Livox-Infra @Livox-SDK @

jfrascon, 2025-09-22:
Hi @shisuhao,

I do not have any experience in openEuler. I checked that platform out on the web, but not sure checking the trace you pasted here what the issue could be.
I can tell you for sure that the livox_ros_driver2 packages compiles in ROS2:humble and in ROS2.jazzy (applying on the PR #180 ) on an ubuntu:22.04 and ubuntu:24:04 Docker images for arm64 with ROS2 humble and jazzy installed on them, respectively.

Jaqen81, 2025-12-03:
I solved this.

1.  Download zip code file, 
2. Paste and unzip it in your ws_livox/src, 
3. Compile it

I met this problem because I copied the source code from Mac computer to Xavier with USB flash drive，there are some ghost files haven't been copied and transmitted. It takes me a day to solve it:) Good Luck